adafruit/Ethernet2 #11: Unnecessarily slow Ethernet interfaces due to none ideal SPI clock

DieterBurandt, 2017-01-18:
On default the W5500 is running with a 8MHz SPI clock with up to 359.712kb/s transfer rate to a PC.

Using an Arduino Due board the W5500 can run with a 28MHz SPI clock with up to 505.500kb/s transfer rate to a PC.

The SPI clock rate depends on the CPU (board), the W5500, and the layout of the board.
Experience user should have a chance to choose his own SPI clock settings.

Please implement a simple method that users can set the SPI speed (e.g. like implemented in SD.cpp already).

Dieter Burandt

ladyada, 2017-02-28:
yeah but a lot of shields have level shifters, if you can do a pull req for setting the SPI speed, ill take a look! :)

DieterBurandt, 2017-02-28:
Hi ladyada,
Sorry, I am not familiar with the Github and how to write a pull request. I didn't find a way to do this.
Six weeks ago I sent Christian my driver and may be, that your current one resides on it. Regarding this topic I will send my current one again with a detailed description about that what I integrated and added. A file compare will help you. I run now my driver on an Arduino Due with 1075.268kb/s (SPI-clock: 28MHz).
[2017-02-28 - Eth2db.zip](https://github.com/adafruit/Ethernet2/files/806932/2017-02-28.-.Eth2db.zip)
Best Regards,
Dieter

DieterBurandt, 2017-03-05:
SI did something wrong with github. It should be still open.

ladyada, 2017-03-06:
hiya we cannot accept a pull request by zip, so we cannot merge this - your best bet is to do a proper PR. thx!

DieterBurandt, 2017-03-07:
Hi  ladyada,
Sorry, I cannot open a new branch and add my files and also I cannot initiate a pull request. I think it is a problem regarding my privileges. Can you help me?  Thanks!

(I compared your Ethernet2 with my one. Merging both together will result into a version that can be immediately published and is full compatible with Arduino 1.8.1. Your one isn’t.)

ladyada, 2017-03-07:
you would create a fork
https://learn.adafruit.com/an-introduction-to-collaborating-with-version-control/

closed 2017-03-07 by ladyada

DieterBurandt, 2017-03-08:
High ladyada,
thank you for this link. Now I understand, what GitHub is and it is clear that I could not open a pull request. I did work over 40 years on hard- and software and for a few years with CVS. But the time passed. I like to help with solutions for bugs and important enhancments. But I don't like to begin working again. I left this world. - Sorry!
My driver is working very well and is based on yours 1.0.3. I did gave my driver to you and I will help you to integrate the updates to your one. You need less then a half hour plus testing. Not more. (Then the driver is much more flexible and more than 5 times faster.) When I try to use GitHub I would need days to support you and finally you need more time for it.

jroberts235, 2017-03-08:
@DieterBurandt, I'd like to offer an hour to bring you up to speed on working in the Github paradigm.   If you're interested, give me you email address and we can try to set something up.  This is not entirely selfless.  I am a user of their library and would like to see the benefits you claim.  I also have some feature requests that you might be able to help me incorporate.  I'm GMT -8

DieterBurandt, 2017-03-08:
Hi J. Roberts,

It is quite late here (GMT+1) in Germany. In around 11 hours I will send 
you a simple cooking list what to implement into the Ethernet2 driver 
1.0.4. (These updates are the same as you can use for the old Ethernet 
W5100 driver as well.)

I include also a test program for that you can use with the Arduino Due 
and with very small changes also with 8 bit CPUs as well. The test 
program "WebServer2-Speedtest" is added as an example.

I add here my driver that works very well. I concatenate the „db“ to the 
file names only to distinguish the version. You don’t need it. Copy the 
files to your library path.

If you don’t have already a powerful file compare program that you like 
very much, download and install the freeware version of Code Compare 4.1 
from www.devart.com. You select two file pathes within the explorer and 
start the program from there. (After starting, you can also mark two 
files with different names within CodeCompare in the two files windows 
using Ctrl + left mouse button and click“Launch Code Compare”.)

By by, till tomorrow

Dieter

DieterBurandt, 2017-03-08:
Hi J. Roberts,

May be that I am in the meantime a little bit confused about the 
versions of the Ethernet files I did use. But the version 1.0.4 of the 
Ethernet2 may be a good base, if it works already without problems.

Do the following:

First:

Look in "Ethernet2db.cpp"  line 17, 19, and 24:
Add this code to the Ethernet2.cpp and the header files to Ethernet2.h.
"SPI_ETHERNET_Clock" is a define in the original Ethernet2.cpp: delete it.
(Now the define is a variable preset to 4MHz. This will be the proper 
value for slow 8-bit CPUs. And because it is now a variable, it can now 
also set by the two function calls. For the Arduino Due e.g. with up to 
42MHz. The official maximum should be 28MHz for the W5500 only but 42MHz 
is also working on my Arduino Due very well.)

Second:

Look in my "W5500.cpp" at line 130 to 131 in "W5500Class::write(...)" 
and at line 163 in "W5500Class::read(...)". Change the code in the 
GitHub file too. The effect is, that no longer a byte transfer call is 
used, but a block transfer call that need significant less code to 
execute. By this, I could enhance the speed by a factor 2.

Third:

  It would be nice, if you would add in Ethernet2.cpp of GitHub also the 
two calls as well and the headers in Ethernet2.h:
IPAddress EthernetClass::dhcpServerIP()
{
      return _dhcp->getDhcpServerIp();
}

void EthernetClass::MacAddr(uint8_t * Mac)
{
       SPI.beginTransaction(SPI_ETHERNET_SETTINGS);
       w5500.getMACAddress(Mac);
       SPI.endTransaction();
}

The two calls are reading the dhcpServerIP and the Mac-address from the 
W5500. This is only for maintenance purpose, but they are missing and 
often importand.

That is all!   ==>  On the Arduino Due: 1160.92kb/s (SPI-clock: 42MHz) 
as tested with "WebServer2-Speedtest"

But there is one point I like to note:

The two calls "SPI.beginTransaction(SPI_ETHERNET_SETTINGS);" and 
"SPI.endTransaction();" are a must, if you use more than one device on 
an SPI bus. If you don't have it, e.g. the speed and mode settings of 
the SD device are also used for the Ethernet device and vice versa. And 
will fail. When I looked today in the GitHub driver, I am not sure if 
all w5500.xxx calls are properly enclosed by the two statements. One 
must not implement the two calls in Ethernet.cpp and in W5500.cpp and 
use it two times for each user call. But if one is completely missing, 
then you have a good chance to come into trouble. And at the same time 
calling it too often has no negative effect despite unnecessary code.

You  can use the  test program "WebServer2-Speedtest" for a speed test. 
The Internet Explorer must support HTML5 download. Not all Internet 
Explorer do support it. Use for example Firefox. If you are testing with 
a Arduino Due you must properly initialise Pin 87 and Pin 77 as an INPUT 
or INPUT_PULLUP.  These two Pins are connected internally to User Pin 4 
and 10 and are initialised by the Arduino software as OUTPUT LOW. That 
are two short circuits: The chip select pins 4 and 20 will never reach a 
full high signal. When I started for about three years with an Arduino 
Due, the SD+Ethernet worked well for two month with low speed and then 
suddenly failed completely. After proper initialisation I had no problem 
anymore.

If you have questions, don't hesitate to ask me.

Best Regards,

Dieter

DieterBurandt, 2017-03-09:
Hi J. Roberts,

I forgot to look at one pont into the driver in GitHub. There is still 
one bug or incomplete implementation in that driver as well as in my one:

In "W5500.h" there are settings of the "W5500 Registers" and "W5500 
Socket registers" as "#define". These settings are code implementations 
and communicate on the SPI-bus.

They must also be encapsulated in 
"SPI.beginTransaction(SPI_ETHERNET_SETTINGS);" and 
"SPI.endTransaction();" as well. I thing it would be the best to move 
the code to the cpp-file and write inline function calls. The machine 
code will be the same as the defines.

Please give my a short resonds, if you got my mails.

Best Regards,

Dieter