adafruit/Ethernet2 #21: Ethernet2 also works with MikroElektronica’s ETH WIZ click

mkarlesky, 2017-09-05:
I can confirm that the Ethernet2 library is also compatible with MikroElektronica’s [ETH WIZ Click](https://shop.mikroe.com/eth-wiz-click) product. It simply requires calling the `init()` method before `begin()` to appropriately set Chip Select to a digital pin routed via the host Click shield. I am working with MikroElektronica’s [MEGA Click shield](https://shop.mikroe.com/arduino-mega-click-shield) with an Arduino Mega. I see no reason why the [UNO Click shield](https://shop.mikroe.com/arduino-uno-click-shield) would not also work.

itsnotyasser, 2018-06-25:
Hi @mkarlesky , Thanks for confirming the compatibility between the ETH Wiz click and the library. 

Have you encountered any connectivity issues when using your Ethernet module? I've noticed mine repeatedly disconnects from the internet.

mkarlesky, 2018-06-25:
@itsnotyasser We've had no trouble with our ETH Wiz Clicks. They've been very reliable with the Ethernet2 library. That said, most of our code does not connect directly to the Internet but to local network resources (via TCP). The one direct-to-Internet communication we do use is SNTP (simple network time protocol). But this is UDP, and UDP is connectionless.

itsnotyasser, 2018-06-25:
Oh yes, should have probably mentioned for myself. Still new to using GitHub. 
This is the second module we've had randomly losing connection. 
(I am continuously pinging the chip via its IP Address, and it seems vibrations make it disconnect.)
I am also using using a UDP with a Communication class based on the Ethernet Library that someone before me had written.

mkarlesky, 2018-06-25:
From our experience so far we feel quite confident in the ETH Wiz Clicks and the Ethernet2 library with them. It's certainly possible that there are bugs or issues we've not yet experienced. Given the circumstances it sounds like the best course of action is to troubleshoot the library you are using as well as the physical setup of your hardware as related to vibration. Without having direct observation of either I don't think I can offer any more recommendations than that. Good luck!

itsnotyasser, 2018-06-25:
Yes, yes of course. Thanks for the feedback.