adafruit/Ethernet2 #32: cannot declare variable 'server' to be of abstract type 'EthernetServer' (ESP32)

Kameeno, 2018-08-02:
Hi!, i'm on platformIO and i'm finding any way to get compiled correctly when i declare `EthernetServer server(80);` 
`src/main.cpp:5:16: error: cannot declare variable 'server' to be of abstract type 'EthernetServer'`

```
#include <Arduino.h>
#include <Ethernet2.h>

EthernetServer server(80);

void setup() {

}

void loop() {
}
```
there's a way to make it run ?
instead on another pc with linux, the same zipped project is working, can you help me :( ?

nnn112358, 2018-09-18:
I have the same problem.

nnn112358, 2018-09-18:
There is no problem with this.

>If new arduino esp32 or make error, need go to c:\Program Files (x86)\Arduino\hardware\espressif\arduino-esp32\cores\esp32\Server.h
>Change virtual void begin(uint16_t port = 0) = 0; to virtual void begin() = 0;

https://github.com/m5stack/M5Stack/blob/master/examples/Modules/W5500/WebServer/WebServer.ino